Jason2866/ESP_Flasher #2: MS Security Essentials finds Trojan:Win32/Zpevdo.B in ESP-Flasher-Windows-x86.exe

AuZ-ASD, 2021-05-04:
Hi,

just downloaded ESP-Flasher-Windows-x86.exe to flash my ESP cam but i cant as MS Security Essentuals blocks and removes because it finds some Trojan:Win32/Zpevdo.B in the file

https://www.microsoft.com/en-us/wdsi/threats/malware-encyclopedia-description?name=Trojan%3aWin32%2fZpevdo.B&threatid=2147729093&enterprise=0

Jason2866, 2021-05-04:
See #1

closed 2021-05-04 by Jason2866

almostlunatic, 2023-11-25:
OK, thanks a lot. Unfortunately _ cannot disable the corporate aantivirus program Bitdefender, so I jhave to find another way tu use it.

Jason2866, 2023-11-25:
Do you have a running Python installed on your PC? Than you can install via pip as described inn the readme

almostlunatic, 2023-11-25:
Nope, but I can do it, thanks!