Jason2866/ESP_Flasher #3: ESP-flasher Mac not running

ApricotFarm, 2021-05-06:
I've downloaded and install both release versions of ESP-flasher to my Mac, but nothing happens when I try to run them. I double-click the icon, the program looks like it's opening, and then just ends, with nothing happening. Does anyone have any idea what's happening?

I'm running macOS 10.13.6 on an Intel MacBook Pro.

Jason2866, 2021-05-06:
I have no Mac. Tested in a VM with Mac Os Catalina. I flashed successfully two devices.
Have the feedback from other Mac users which have successfully flashed devices

ApricotFarm, 2021-05-07:
Hmm. Do you know if the other users using 10.13, or were all using newer OS versions? 

I can try loading up the Linux version -- it would be Ubuntu on a RasPi 4. Is there any reason to think that wouldn't work?

Jason2866, 2021-05-07:
Positive Feedback was with Mac Os Catalina 10.15.7

dweston, 2021-05-20:
I have the same problem as @ApricotFarm although my MBP is running macOS Mohave Version 10.14.6.

dweston, 2021-05-20:
I tried the `Build it yourself` path and all seemed OK until I called the `esp_flasher` app.

Result was:
`This program needs access to the screen. Please run with a
Framework build of python, and only when you are logged in
on the main display of your Mac.
`
Which lead to this:
[https://stackoverflow.com/questions/48531006/wxpython-this-program-needs-access-to-the-screen](url)

Which lead to this:
[https://wiki.wxpython.org/wxPythonVirtualenvOnMac](url)

However I'm not sufficiently `python` savvy to know what to do from here.

It appears it is necessary to start GUI scripts with `pythonw` as opposed to `python`.

Does anyone know how to kick the `esp_flasher` app off as a python script?

dweston, 2021-05-22:
Looks like `XCode` has to be installed to `Build it yourself`. 
[https://realpython.com/python-gui-with-wxpython/#installing-wxpython](url)

Jason2866, 2021-05-22:
As written in `readme`

ApricotFarm, 2021-05-22:
Hi @dweston -- I've finally had some success with this, using the "build it yourself" method on a Raspberry Pi (you DO have a Raspberry Pi sitting around, right? :-) It was a bit of a long walk, but I'll post more here once I've had the chance to consolidate the information. Although...it might be time to close this particular thread/issue, and open a separate one related to different methods of getting this running? @Jason2866, what do you think?

Jason2866, 2021-05-23:
Closing since it is not a ESP-Flasher issue.

dweston, 2021-05-23:
I got it working via command line invocation. Followed some [tutorials](https://realpython.com/python-gui-with-wxpython/#installing-wxpython) on creating a skeleton GUI app usig wxpython, which worked. Not sure why but it now does what I need it to do on Mohave.

Jason2866, 2021-09-24:
Uploaded a MacOS BigSur build. Compiled on MacBook Air M1 with Rosetta. It does work on my M1 and it should work on x86 Macs too since it is a x86 processor build.

ApricotFarm, 2021-09-24:
Thanks @ Jason2866. I'm still hoping to get a working version for my High Sierra installation (10.13.6). @dweston: would you be willing to post your working Mohave build? I'm guessing there's a good chance that it will work on my setup.

On a separate note, I have been **unable** to replicate my success building a working version on Ubuntu Hirsute on a RasPi 4, which is why I never followed up on the precise steps I took to get the build working. I did, however, also manage to get the Windows 10 version running on Raspberry Pi, using @botspot's amazing script (which can be found here: https://github.com/Botspot/wor-flasher) and have successfully flashed a few ESP32 chips without issue.

Jason2866, 2021-09-25:
Have you tried to start the App a second time, just right after the first when nothing happened?

ApricotFarm, 2021-09-25:
Hi, I'm not sure if you're addressing that to me, but if so, yes...I tried it many times (I haven't really touched the mac version since May), and just tried it a few time right now, just for good measure. Nuthin'.

> Have you tried to start the App a second time, just right after the first when nothing happened?