Jason2866/ESP_Flasher #6: Flashing ESP32 fails with no explicit error

DanChaltiel, 2021-06-23:
I bought an ESP-30 CAM device with the built-in USB interface (on [aliexpress](https://fr.aliexpress.com/item/1005001997158739.html?spm=a2g0s.9042311.0.0.7b046c37NS2Sey), seems to be on [amazon](https://www.amazon.fr/Dreamitpossible-ESP32-CAM-CH340-d%C3%A9veloppement-cam%C3%A9ra-Bluetooth/dp/B07YBPLQ4B) too):

[image]

I downloaded `tasmota32-webcam.bin` from http://ota.tasmota.com/tasmota32/release/ and flashed it to my device using ESP Flasher.

Unlike many tutorials I saw, I had to press the "Flash" hardware button while pressing "Flash ESP" on the software, but I had the same results by wiring IO0 to GND.

Here is the output. As you can see, the logs don't show:

<details>

  <summary>ESP-flasher output</summary>

```
Using 'COM5' as serial port.
Connecting........
Detecting chip type... ESP32
Connecting...

Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32-D0WDQ6 (revision 1)
 - Number of Cores: 2
 - Max CPU Frequency: 240MHz
 - Has Bluetooth: YES
 - Has Embedded Flash: NO
 - Has Factory-Calibrated ADC: YES
 - MAC Address: 10:52:1C:64:32:D8
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
 - Flash Size: 4MB
 - Flash Mode: qio
 - Flash Frequency: 80MHz
Erasing flash (this may take a while)...
Chip erase completed successfully in 2.7s
Flash params set to 0x002f
Compressed 18656 bytes to 12053...
Writing at 0x00001000... (100 %)
Wrote 18656 bytes (12053 compressed) at 0x00001000 in 0.3 seconds (effective 549.3 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 129...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (129 compressed) at 0x00008000 in 0.0 seconds (effective 4034.9 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 15296.5 kbit/s)...
Hash of data verified.
Compressed 1425152 bytes to 832040...
Writing at 0x00010000... (1 %)
Writing at 0x00014000... (3 %)
Writing at 0x00018000... (5 %)
Writing at 0x0001c000... (7 %)
Writing at 0x00020000... (9 %)
Writing at 0x00024000... (11 %)
Writing at 0x00028000... (13 %)
Writing at 0x0002c000... (15 %)
Writing at 0x00030000... (17 %)
Writing at 0x00034000... (19 %)
Writing at 0x00038000... (21 %)
Writing at 0x0003c000... (23 %)
Writing at 0x00040000... (25 %)
Writing at 0x00044000... (27 %)
Writing at 0x00048000... (29 %)
Writing at 0x0004c000... (31 %)
Writing at 0x00050000... (33 %)
Writing at 0x00054000... (35 %)
Writing at 0x00058000... (37 %)
Writing at 0x0005c000... (39 %)
Writing at 0x00060000... (41 %)
Writing at 0x00064000... (43 %)
Writing at 0x00068000... (45 %)
Writing at 0x0006c000... (47 %)
Writing at 0x00070000... (49 %)
Writing at 0x00074000... (50 %)
Writing at 0x00078000... (52 %)
Writing at 0x0007c000... (54 %)
Writing at 0x00080000... (56 %)
Writing at 0x00084000... (58 %)
Writing at 0x00088000... (60 %)
Writing at 0x0008c000... (62 %)
Writing at 0x00090000... (64 %)
Writing at 0x00094000... (66 %)
Writing at 0x00098000... (68 %)
Writing at 0x0009c000... (70 %)
Writing at 0x000a0000... (72 %)
Writing at 0x000a4000... (74 %)
Writing at 0x000a8000... (76 %)
Writing at 0x000ac000... (78 %)
Writing at 0x000b0000... (80 %)
Writing at 0x000b4000... (82 %)
Writing at 0x000b8000... (84 %)
Writing at 0x000bc000... (86 %)
Writing at 0x000c0000... (88 %)
Writing at 0x000c4000... (90 %)
Writing at 0x000c8000... (92 %)
Writing at 0x000cc000... (94 %)
Writing at 0x000d0000... (96 %)
Writing at 0x000d4000... (98 %)
Writing at 0x000d8000... (100 %)
Wrote 1425152 bytes (832040 compressed) at 0x00010000 in 18.8 seconds (effective 607.0 kbit/s)...
Hash of data verified.

Leaving...
Hard Resetting...
Done! Flashing is complete!

Showing logs:
```

</details>

If I click on "View Logs", I get this error:

```
Using 'COM5' as serial port.
Unexpected error: could not open port 'COM5': PermissionError(13, 'Accès refusé.', None, 5)
```

Of course, there was never any Tasmota hotspot available, I checked at every step.

If I reset the device (without IO0 wired to GND), I get logs that are sometimes unreadable (like if the baudrate was off) and the device seems to reboot every second. I can obviously not check the output with a better power source than my computer USB, but the device seems to reboot with a powerful (3A) source as well, and it doesn't provide more wifi hotspot.

Here are an example of (readable) log output:

<details>
  <summary>Logs</summary>
```
[17:55:15]ets Jun  8 2Ғ:57
[17:55:15]
[17:55:15]rst:0x10 (*]Q_RTC_RESET),boot:03 (SPI_FAST_FLASH_BOOT)
[17:55:15]˥sip: 0, SPIWP:0xee
[17:55:15]ckE:0x00,q_drv:0x00,d_drv:0xb_drv:0x00,hd_drv:0x00,wp_dr0
[17:55:15]mode:QIO, clock div:1
[17:55:15]l+0x3fff0018,len:4
[17:55:15]load:0xN&121,len:-1725455839
[17:55:15]1162 mmu set 00010000, pk00010000
[17:55:15]1162 mmu set 0002b́00020000
[17:55:15]1162 mmu setb pos 00030000
[17:55:15]116&j set 00040000, pos 0
[17:55:15]1162 mmu set 00050000, po.0
[17:55:15]1162 mmu set 00060000, ́00060000
[17:55:15]1162 mmu set 00070000, pos
[17:55:15]1162 mmu set 00080000, po.00
[17:55:15]1LjՁset 00090000, pos 000Nj
[17:55:15]1162 mmu set 000a0000,́000a0000
[17:55:15]1162 mmu set 000b0000, po.000
[17:55:15]1162 mmu set 000c0000, pos000
[17:55:15]1162 mm]с000d0000, pos 000d000j
[17:55:15]1162 mmu set*000, pos 000e0000
[17:55:15]1162 mmu 2000, pos 000f0000
```
</details>

I tried with different cables and different COM ports with the same effect.

Is there any chance this might come from ESP Flasher?

Jason2866, 2021-06-23:
Power issue.

closed 2021-06-23 by Jason2866

DanChaltiel, 2021-06-25:
Wow, that was a bit harsh of you, especially as I definitely think it is not a power issue.

Using the exact same hardware setting (I even changed my USB to TTL adapter to be sure), I could transfer the [esp32cam example](https://dl.espressif.com/dl/package_esp32_index.json
) using the arduino IDE and it worked pretty fine, I could stream my video neatly. 
Here is the tutorial that I used: https://projetsdiy.fr/esp32-cam-aithinker-flash-firmware-test/ (in french but google translate will do).

Also, it seems I'm not the only one with a problem with ESP_flasher: https://www.reddit.com/r/homeassistant/comments/cqe705/could_not_open_port_com5_permissionerror13_access/

This either comes from ESP_flasher or from Tasmota but I'm rather sure it does not comes from a hardware issue.

Jason2866, 2021-06-25:
Sorry for being so short. The error log you posted is the typical one for a power issue.
I still think it is a hardware problem. The problem you may have is, your board does not support the max performance settings choosen in Tasmota to get best and fastest video streams
Which settings for CPU clock, Flash mode, Flash clock do you have for the test sketch?
Tasmota uses: CPU 240Mhz, Flash mode QIO, Flash clock 80Mhz. This are the highest allowed settings and are recommended for the AI tinker esp32 webcam. Mine works fine with these settings flashed with ESP_flasher.
If lower settings are choosen/needed it is a bad (cheap) board design

adrienzagar, 2021-06-30:
Hey ! I had the exact same problem as you DanChaltiel, with the same hardware, solved by upgrading the firmware tasmota32 to tasmota32-webcam from the web interface !! 
Tip that probably save me a lot of time !! Thank you !

raspitoaster, 2021-07-06:
Thanks for the tip - I flashed 4 identical ESPcams and only one had this boot loop after flashing tasmota-webcam binary.
I was able to recover this beast by flashing first tasmota.bin and then tasmota-webcam.bin