Jason2866/ESP_Flasher #7: Bad flash, tons of log spam, etc

codefaux, 2021-07-09:
Hi there. Just downloaded and used v1.1 (I think, there's no clear version indication on the software anywhere I can find) on a factory-new ESP32-Cam module with a freshly downloaded tasmota32-webcam.bin firmware.

It seemed to do the usual work, but after finishing the Console window was scrolling VOLUMES of random characters occasionally tagged with a timestamp and sometimes a few other lines I couldn't read. I couldn't stop the scrolling lines; they were CONSTANTLY churning past.

The only way I could find to interrupt it was to unplug the device, which caused the application to instantly crash and disappear.

I tried again a second time, but this time it was vomiting text I couldn't read (or log, or stop, or slow down) over and over and over and over and over .....so I unplugged the device again, and tried a third time.

Same results.

After that, I manually downloaded all of the files from the Tasmota Github and manually flashed it using the following command;
`esptool.py --chip esp32 --port COM9 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dout --flash_freq 40m --flash_size detect 0x1000 bootloader_dout_40m.bin 0x8000 partitions.bin 0xe000 boot_app0.bin 0x10000 tasmota32-webcam.bin`

Manually flashing worked fine. Reflashing with ESP_Flasher AFTER manually flashing results in the exact same behavior -- repeated, scrolling, spamming text that goes past so fast I can't read it.

I'd love to give more information, but the application doesn't leave logs behind to examine.

codefaux, 2021-07-09:
Oh hey this time it didn't crash instantly when I unplugged the device in a desperate attempt to make the application stop spamming and close. Here you go.

`Using 'COM9' as serial port.
Connecting.....
Detecting chip type... ESP32
Connecting....

Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32-D0WDQ6 (revision 1)
 - Number of Cores: 2
 - Max CPU Frequency: 240MHz
 - Has Bluetooth: YES
 - Has Embedded Flash: NO
 - Has Factory-Calibrated ADC: YES
 - MAC Address: 84:CC:A8:7B:9D:C8
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
 - Flash Size: 4MB
 - Flash Mode: qio
 - Flash Frequency: 80MHz
Erasing flash (this may take a while)...
Chip erase completed successfully in 3.2s
Flash params set to 0x002f
Compressed 18656 bytes to 12053...
Writing at 0x00001000... (100 %)
Wrote 18656 bytes (12053 compressed) at 0x00001000 in 0.3 seconds (effective 550.3 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 129...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (129 compressed) at 0x00008000 in 0.0 seconds (effective 4436.6 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 21705.5 kbit/s)...
Hash of data verified.
Compressed 1425152 bytes to 832040...
Writing at 0x00010000... (1 %)
Writing at 0x00014000... (3 %)
Writing at 0x00018000... (5 %)
Writing at 0x0001c000... (7 %)
Writing at 0x00020000... (9 %)
Writing at 0x00024000... (11 %)
Writing at 0x00028000... (13 %)
Writing at 0x0002c000... (15 %)
Writing at 0x00030000... (17 %)
Writing at 0x00034000... (19 %)
Writing at 0x00038000... (21 %)
Writing at 0x0003c000... (23 %)
Writing at 0x00040000... (25 %)
Writing at 0x00044000... (27 %)
Writing at 0x00048000... (29 %)
Writing at 0x0004c000... (31 %)
Writing at 0x00050000... (33 %)
Writing at 0x00054000... (35 %)
Writing at 0x00058000... (37 %)
Writing at 0x0005c000... (39 %)
Writing at 0x00060000... (41 %)
Writing at 0x00064000... (43 %)
Writing at 0x00068000... (45 %)
Writing at 0x0006c000... (47 %)
Writing at 0x00070000... (49 %)
Writing at 0x00074000... (50 %)
Writing at 0x00078000... (52 %)
Writing at 0x0007c000... (54 %)
Writing at 0x00080000... (56 %)
Writing at 0x00084000... (58 %)
[... 4122 lines cut ...]
[17:27:03]1162 mmu set 000b0000, pos 000b0000
[17:27:03]1162 mmu set 000c0000, pos 000c0000
[17:27:03]1162 mmu set 000d0000, pos 000d0000
[17:27:03]1162 mmu set 000e0000, pos 000e0000
[17:27:03]1162 mmu set 000f0000, pos 000f0000
[17:27:04]1162 mmu set 00100000, pos 00100000
[17:27:04]1162 mmu set 00110000, pos 00110000
[17:27:04]ets Jun  8 2016 00:22:57
[17:27:04]
[17:27:04]rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[17:27:04]configsip: 0, SPIWP:0xee
[17:27:04]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[17:27:04]mode:QIO, clock div:1
[17:27:04]load:0x3fff0018,len:4
[17:27:04]load:0x928b2121,len:-1725455839
[17:27:04]1162 mmu set 00010000, pos 00010000
[17:27:04]1162 mmu set 00020000, pos 00020000
[17:27:04]1162 mmu set 00030000, pos 00030000
[17:27:04]1162 mmu set 00040000, pos 00040000
[17:27:04]1162 mmu set 00050000, pos 00050000
[17:27:04]1162 mmu set 00060000, pos 00060000
[17:27:04]1162 mmu set 00070000, pos 00070000
[17:27:04]1162 mmu set 00080000, pos 00080000
Serial port closed!
`

Jason2866, 2021-07-09:
Flashing manually you used mode DOUT and flash frequency 40 Mhz.
The standard settings for Webcam build are flash mode QIO (faster) and flash frequency 80 Mhz (again faster). Your webcam does not statisfy the standards which a original AI-Thinker Webcam has. **Or** the power supply is not stable enough to power the webcam with the faster settings which draw more current

codefaux, 2021-07-09:
To be clear, I just copy-pasted the command from the Tasmota documentation to flash the device, so it was less my idea, but point taken -- your app chooses settings which aren't compatible with some configurations, whereas the Tasmota documentation provides working defaults for most cases (even potentially unstable power supplies and/or devices which don't satisfy the 'standards' set by another device) thus providing a stable experience at the expense of a little speed.

I'll keep that in mind going forward and just do it manually, thanks a ton.

Just seemed odd, since the logs claimed to have done a data hash verification, seems like that might detect such a problem, lol. Sounds like they should patch the verification routine. Is there some way you can detect that issue and prevent the hard-infinite-loop from happening, at least? "If process X respawns more than ten thousand times in thirty seconds, something went wrong and maybe let's stop doing that" or something along those lines?

Also sorry for excessive pages of logcrap, I put it in the Code formatting but I didn't realize it was so long it was going to break that.

Jason2866, 2021-07-09:
This log spam in first step can be a bad power supply. Most users are powering the device with the USB-serial adapter when flashing. With a good Usb serial adapter this is okay. But many ESP32 devices will bootloop when trying to start. You will see the garbage you had. In most cases the flashed device will work when powered correctly.

Jason2866, 2021-07-09:
About flashing we (the Tasmota dev team) are working to port the ESP Webflasher to be used with Tasmota. So flashing needs only Chrome or Edge browser (and a USB serial adapter of course) to flash ALL Tasmota variants just with a click in Browser

JustSpring, 2021-07-09:
Hi, I have the same problem. the esp32 is connected directly to the USB on the computer so I don't think it has a power problem. Do you have something I can do to fix it? Thanks

> This log spam in first step can be a bad power supply. Most users are powering the device with the USB-serial adapter when flashing. With a good Usb serial adapter this is okay. But many ESP32 [...]

Jason2866, 2021-07-09:
First working version of Tasmota Webflasher. The flashing functions are final. 
Only the different Firmware variants are missing. I added Tasmota32-webcam already. 
You can test if you want. It is here [https://jason2866.github.io/Tasmota/](https://jason2866.github.io/Tasmota/)

Jason2866, 2021-07-10:
Issue is probably generated by this https://github.com/arendst/Tasmota/pull/12613
It is not a ESP_Flasher issue. It does what is told in the firmware files.

clementcaen, 2021-07-10:
Thanks a lot Jason2866, I have the same problem and I don't know why but I trying with the web solution and it works ! 
Since two days I try different solution, I install ubuntu, debian, try manually, ... and finally achieve with this simple click... 😅

Jason2866, 2021-07-10:
We are heavily working on this tool. At the moment the webcam build is "out".
I will add it again tomorrow. It is in early alpha state.
We will announce when it is ready for daily use. There is some work to do...

Jason2866, 2021-07-24:
Closing since no feedback using modified Tasmota webcam build (dout now default).

closed 2021-07-24 by Jason2866