Jason2866/ESP_Flasher #14: Missing file api-ms-win-core-path-l1-1-0.dll when starting ESP-Flasher-Windows-x86.exe

Sum-Guy, 2022-01-20:
I'm trying to run ESP-Flasher-Windows-x86.exe on a win-7 32-bit PC and I immediately get the message "The program can't start because api-ms-win-core-path-l1-1-0.dll is missing from your computer."  Using the program "Everything" to search for files, I see that I have many "api-ms-win-core-" files but no "api-ms-win-core-path-" files.  When I dismiss the error message a second message comes up - Error loading Python DLL - and then it shows a path to a file called python39.dll.  That file does indeed exist at the given path.  The PC in question does have python 3.7 installed (I believe it is the last version that will run on windows 7) but I don't know if that's a factor here.

Sum-Guy, 2022-01-20:
Edit:  I found this:
https://vigilance.fr/vulnerability/Python-executing-DLL-code-via-Windows-7-api-ms-win-core-path-l1-1-0-dll-31456

I don't know if that is useful.  I have also downloaded and run vc_redist.x86.exe (Microsoft Visual C++ Redistributable) but I still don't seem to have that DLL file.  I have come across some indication that the DLL file first appeared in Windows 8.  So it may not exist for Windows 7?

Jason2866, 2022-01-20:
Thanks for the info. I will remove the x86 version.

closed 2022-01-20 by Jason2866

Sum-Guy, 2022-01-20:
Well, I didn't want to see the x86 version removed...

Jason2866, 2022-01-20:
It is not possible to support anymore. There is no github actions build env which can be used to build a working x86 variant.