Jason2866/ESP_Flasher #19: Problem flashing Shelly plus 1PM

FerFemNemBem, 2022-05-02:
Hi,
I try to flash a "Shelly plus 1PM" with "tasmota32solo1.bin".
"Something" seems start and the 1PM is recognized but then it fails (everytime at the same place):

```
Using 'COM6' as serial port.
Connecting.....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
Connecting....

Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32-U4WDH (revision 3)
 - Number of Cores: 1
 - Max CPU Frequency: 80MHz
 - Has Bluetooth: YES
 - Has Embedded Flash: YES
 - Has Factory-Calibrated ADC: YES
 - MAC Address: 7C:87:CE:64:BA:94
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Chip does not support baud rate 460800, changing to 115200
Connecting......................................
Unexpected error: ESP Chip Auto-Detection failed: Failed to connect to Espressif device: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
```

Any idea what I'm doing wrong here?

BR,
FFNB

FerFemNemBem, 2022-05-02:
Managed in the meantime to flash 1 of my 2 devices using the Webinstaller ( https://tasmota.github.io/install/ ).
Left the other one untouched if I should test something for this issue here with it.

Jason2866, 2022-05-02:
Can not reproduce. Does work.
```
Using '/dev/cu.SLAB_USBtoUART' as serial port.
Connecting.....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting.....
Detecting chip type... ESP32
Connecting....

Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32-S0WDQ5 (revision 1)
 - Number of Cores: 1
 - Max CPU Frequency: 80MHz
 - Has Bluetooth: YES
 - Has Embedded Flash: NO
 - Has Factory-Calibrated ADC: YES
 - MAC Address: 7C:9E:BD:35:F2:E4
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
 - Flash Size: 4MB
 - Flash Mode: dout
 - Flash Frequency: 40MHz
Erasing flash (this may take a while)...
Chip erase completed successfully in 16.5s
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x00129fff...
Compressed 17088 bytes to 11846...
Writing at 0x00001000... (100 %)
Wrote 17088 bytes (11846 compressed) at 0x00001000 in 0.6 seconds (effective 242.8 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 129...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (129 compressed) at 0x00008000 in 0.1 seconds (effective 353.3 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 566.9 kbit/s)...
Hash of data verified.
Compressed 1154688 bytes to 806009...
Writing at 0x00010000... (2 %)
Writing at 0x000168e4... (4 %)
Writing at 0x00020b73... (6 %)
Writing at 0x00028e0c... (8 %)
Writing at 0x000309b7... (10 %)
Writing at 0x00038c3e... (12 %)
Writing at 0x00041cdc... (14 %)
Writing at 0x00046a44... (16 %)
Writing at 0x0004bd2e... (18 %)
Writing at 0x00050bf8... (20 %)
Writing at 0x00055f10... (22 %)
Writing at 0x0005b98f... (24 %)
Writing at 0x00060cc5... (26 %)
Writing at 0x00066837... (28 %)
Writing at 0x0006bc6e... (30 %)
Writing at 0x00070d2f... (32 %)
Writing at 0x000764b4... (34 %)
Writing at 0x0007b9ea... (36 %)
Writing at 0x00080973... (38 %)
Writing at 0x0008581a... (40 %)
Writing at 0x0008a6f1... (42 %)
Writing at 0x0008f4e3... (44 %)
Writing at 0x0009409c... (46 %)
Writing at 0x00099db3... (48 %)
Writing at 0x0009f40a... (50 %)
Writing at 0x000a4e8d... (52 %)
Writing at 0x000aa258... (54 %)
Writing at 0x000af5e7... (56 %)
Writing at 0x000b4ae3... (58 %)
Writing at 0x000b9d8f... (60 %)
Writing at 0x000bf216... (62 %)
Writing at 0x000c453c... (64 %)
Writing at 0x000c9446... (66 %)
Writing at 0x000ce767... (68 %)
Writing at 0x000d446f... (70 %)
Writing at 0x000d9a88... (72 %)
Writing at 0x000decd2... (74 %)
Writing at 0x000e40b3... (76 %)
Writing at 0x000e95d9... (78 %)
Writing at 0x000eec32... (80 %)
Writing at 0x000f43e5... (82 %)
Writing at 0x000fa1de... (84 %)
Writing at 0x000ffae8... (86 %)
Writing at 0x00104e48... (88 %)
Writing at 0x0010a817... (90 %)
Writing at 0x0010fa4a... (92 %)
Writing at 0x001180dc... (94 %)
Writing at 0x0011dc19... (96 %)
Writing at 0x00122f49... (98 %)
Writing at 0x00128bc3... (100 %)
Wrote 1154688 bytes (806009 compressed) at 0x00010000 in 19.4 seconds (effective 475.9 kbit/s)...
Hash of data verified.

Leaving...
Hard Resetting...
Hard resetting via RTS pin...
Done! Flashing is complete!

Showing logs:
[16:24:40]
[16:24:40]00:00:00.003 HDW: ESP32-S0WD 
[16:24:40]./components/esp_littlefs/src/littlefs/lfs.c:1071:error: Corrupted dir pair at {0x0, 0x1}
[16:24:41]00:00:00.487 UFS: FlashFS mounted with 312 kB free
[16:24:41]00:00:00.491 CFG: Use defaults
```

FerFemNemBem, 2022-05-02:
Hmm, ok.
as the webinstall is a workaround - should we (or you) invest time here or should I close?

BR,
FFNB

Jason2866, 2022-05-02:
Fallback to speed 115200 is a indicator of a bad connection or weak power supply.

I see nothing what can cause your issue in the program. Since i refactored the webserial part of the WebFlasher i knew both. Tbh i am surprised webserial does work better.
The "engine" is a minimalistic implementation of esptool.py.
Esp_Flasher has the origi. esptool.py v3.3 "under the hood"

FerFemNemBem, 2022-05-03:
I've tried right nows with my second Shelly 1 Plus PM.
Again the error on using ESP_Flasher. The Webinstall was successfull.

Best regards,
FFNB

closed 2022-05-03 by FerFemNemBem