Jason2866/ESP_Flasher #23: Ability to flash on a off line computer

hallard, 2023-01-14:
Hi @Jason2866, I got some issues on production factory that have computers not connected to Internet or with some strange Firewall rules that prevent downloading bootloader

[image]

Unexpected error: Error while retrieving firmware file 'https://github.com/espressif/arduino-esp32/raw/2.0.3-RC1/tools/sdk/esp32/bin/bootloader_dio_40m.bin': HTTPSConnectionPool(host='[github.com](http://github.com/)', port=443): Max retries exceeded with url: /espressif/arduino-esp32/raw/2.0.3-RC1/tools/sdk/esp32/bin/bootloader_dio_40m.bin (Caused by SSLError(SSLEOFError(8, 'EOF occurred in violation of protocol (_ssl.c:1129)')))

Looks like Chineese inspection SSL firewall

Anyway any option to use a local file that I can provide to factory ?

Thanks

Jason2866, 2023-01-14:
@hallard Yes, using the generated factory.image firmware. Simply flash via esptool.py or any other tool which is capable to flash at address 0x0 and do NOT modify the firmware file.

Jason2866, 2023-01-14:
I am open for a PR implementing flashing a factory image for the ESP32x MCUs :-)
Tbh, I don't invest time to enhance my tool, made it long before the nice way via WebInstaller was there. I do find it deprecated this days...

hallard, 2023-01-14:
>Tbh, I don't invest time to enhance my tool

Totally understand, don't worry.
What I love in your tool is flashing + immediate seeing log on console, I like seeing what's going on on background, and see tasmota is starting good :-)

We will try another way with windows tool, but I'm on MAC I need to find a windows to write a procedure, in this case if I understand we need to flash "factory" image starting a address 0x00 correct?

Jason2866, 2023-01-14:
Yes, all factory images need to be flashed at 0x0

Jason2866, 2023-03-17:
Closing since not planned to implement

closed 2023-03-17 by Jason2866

Jason2866, 2023-11-17:
With newest version which supports factory images, it should work offline.