Jason2866/ESP_Flasher #24: Installing via pip fails

jeroenst, 2023-03-17:
Hi!

I try to use ESP_Flasher but I get an error while installing via pip:

``
pip install esp-flasher
Defaulting to user installation because normal site-packages is not writeable
Collecting esp-flasher
  Using cached esp_flasher-1.4.1-py3-none-any.whl (111 kB)
Requirement already satisfied: requests<3,>=2.24.0 in /home/jeroen/.local/lib/python3.8/site-packages (from esp-flasher) (2.27.1)
Collecting wxpython<5.0,>=4.0
  Using cached wxPython-4.2.0.tar.gz (71.0 MB)
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [8 lines of output]
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "/tmp/pip-install-zyftjlzj/wxpython_11e66a7a8d8f422ab83c39fad694ad67/setup.py", line 27, in <module>
          from buildtools.config import Config, msg, opj, runcmd, canGetSOName, getSOName
        File "/tmp/pip-install-zyftjlzj/wxpython_11e66a7a8d8f422ab83c39fad694ad67/buildtools/config.py", line 30, in <module>
          from attrdict import AttrDict
      ModuleNotFoundError: No module named 'attrdict'
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
``

Jason2866, 2023-03-17:
Which OS and version?

jeroenst, 2023-03-17:
Linux Mint 20.3 Cinnamon

jeroenst, 2023-03-17:
pip3 install attrdict solves this error but now I got this:

pip install esp-flasher
Defaulting to user installation because normal site-packages is not writeable
Collecting esp-flasher
  Using cached esp_flasher-1.4.1-py3-none-any.whl (111 kB)
Requirement already satisfied: requests<3,>=2.24.0 in ./.local/lib/python3.8/site-packages (from esp-flasher) (2.27.1)
Requirement already satisfied: pyserial>=3.5 in ./.local/lib/python3.8/site-packages (from esp-flasher) (3.5)
Collecting wxpython<5.0,>=4.0
  Using cached wxPython-4.2.0.tar.gz (71.0 MB)
  Preparing metadata (setup.py) ... done
Requirement already satisfied: urllib3<1.27,>=1.21.1 in ./.local/lib/python3.8/site-packages (from requests<3,>=2.24.0->esp-flasher) (1.26.5)
Requirement already satisfied: idna<4,>=2.5 in /usr/lib/python3/dist-packages (from requests<3,>=2.24.0->esp-flasher) (2.8)
Requirement already satisfied: charset-normalizer~=2.0.0 in ./.local/lib/python3.8/site-packages (from requests<3,>=2.24.0->esp-flasher) (2.0.12)
Requirement already satisfied: certifi>=2017.4.17 in /usr/lib/python3/dist-packages (from requests<3,>=2.24.0->esp-flasher) (2019.11.28)
Requirement already satisfied: pillow in /usr/lib/python3/dist-packages (from wxpython<5.0,>=4.0->esp-flasher) (7.0.0)
Requirement already satisfied: six in ./.local/lib/python3.8/site-packages (from wxpython<5.0,>=4.0->esp-flasher) (1.16.0)
Requirement already satisfied: numpy in ./.local/lib/python3.8/site-packages (from wxpython<5.0,>=4.0->esp-flasher) (1.24.2)
Building wheels for collected packages: wxpython
  Building wheel for wxpython (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [142 lines of output]
      running bdist_wheel
      running build
      WARNING: Building this way assumes that all generated files have been
      generated already.  If that is not the case then use build.py directly
      to generate the source and perform the build stage.  You can use
      --skip-build with the bdist_* or install commands to avoid this
      message and the wxWidgets and Phoenix build steps in the future.
      
      "/usr/bin/python3" -u build.py build
      Will build using: "/usr/bin/python3"
      3.8.10 (default, Nov 14 2022, 12:59:47)
      [GCC 9.4.0]
      Python's architecture is 64bit
      cfg.VERSION: 4.2.0
      
      Running command: build
      Running command: build_wx
      wxWidgets build options: ['--wxpython', '--unicode', '--gtk3']
      Configure options: ['--enable-unicode', '--with-gtk=3', '--enable-sound', '--enable-graphics_ctx', '--enable-display', '--enable-geometry', '--enable-debug_flag', '--enable-optimise', '--disable-debugreport', '--enable-uiactionsim', '--enable-autoidman', '--with-sdl']
      /tmp/pip-install-x1sgfktd/wxpython_b9652ea1547946a393162e6a5fea9a72/ext/wxWidgets/configure --enable-unicode --with-gtk=3 --enable-sound --enable-graphics_ctx --enable-display --enable-geometry --enable-debug_flag --enable-optimise --disable-debugreport --enable-uiactionsim --enable-autoidman --with-sdl
      checking build system type... x86_64-pc-linux-gnu
      checking host system type... x86_64-pc-linux-gnu
      checking for toolkit... gtk
      checking for gcc... gcc
      checking whether the C compiler works... yes
      checking for C compiler default output file name... a.out
      checking for suffix of executables...
      checking whether we are cross compiling... no
      checking for suffix of object files... o
      checking whether we are using the GNU C compiler... yes
      checking whether gcc accepts -g... yes
      checking for gcc option to accept ISO C89... none needed
      checking whether we are using the Intel C compiler... no
      checking how to run the C preprocessor... gcc -E
      checking for grep that handles long lines and -e... /usr/bin/grep
      checking for egrep... /usr/bin/grep -E
      checking whether gcc needs -traditional... no
      checking for g++... g++
      checking whether we are using the GNU C++ compiler... yes
      checking whether g++ accepts -g... yes
      checking whether we are using the Intel C++ compiler... no
      checking whether we are using the Sun C++ compiler... no
      checking for ar... ar
      checking for ANSI C header files... yes
      checking for sys/types.h... yes
[... 237 lines cut ...]
      --libs' are in the LD_LIBRARY_PATH or equivalent.
      
      Error running configure
      ERROR: failed building wxWidgets
      Traceback (most recent call last):
        File "build.py", line 1563, in cmd_build_wx
          wxbuild.main(wxDir(), build_options)
        File "/tmp/pip-install-x1sgfktd/wxpython_b9652ea1547946a393162e6a5fea9a72/buildtools/build_wxwidgets.py", line 379, in main
          exitIfError(wxBuilder.configure(dir=wxRootDir, options=configure_opts),
        File "/tmp/pip-install-x1sgfktd/wxpython_b9652ea1547946a393162e6a5fea9a72/buildtools/build_wxwidgets.py", line 72, in exitIfError
          raise builder.BuildError(msg)
      buildtools.builder.BuildError: Error running configure
      Finished command: build_wx (0m3.703s)
      Finished command: build (0m3.703s)
      Command '"/usr/bin/python3" -u build.py build' failed with exit code 1.
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: legacy-install-failure

× Encountered error while trying to install package.
╰─> wxpython

note: This is an issue with the package mentioned above, not pip.
hint: See above for output from the failure.

Jason2866, 2023-11-03:
Can you try latest version?

jeroenst, 2023-11-09:
pip install esp-flasher gives the same error. Do I need to execute another command to install the latest version?

Jason2866, 2023-11-09:
The problem is installing wxpython. It is the gfx module dependency. On which Linux platform is Linux Mint 20.3 Cinnamon based?

Jason2866, 2023-11-09:
this prerequisites 
```
sudo apt install python3-pip make gcc libgtk-3-dev libgstreamer-gl1.0-0 freeglut3 freeglut3-dev python3-gst-1.0 libglib2.0-dev ubuntu-restricted-extras libgstreamer-plugins-base1.0-dev
```
and after that it should be possible to build wxPython
```
pip3 install -v --user wxPython
```

Jason2866, 2023-11-09:
When wxpython is successful installing ESP Flasher should work.

jeroenst, 2024-01-02:
Sorry that it took so long before my reply, but I can confirm your prerequisites works for linux mint correctly.

I can now run esp flasher.

Thank you!

closed 2024-01-02 by jeroenst