Jason2866/ESP_Flasher #27: Cannot flash tasmota32.bin v13.2.0

felipeapp, 2023-10-27:
The esp32 keeps rebooting after flashing this binary (v13.2.0):
https://ota.tasmota.com/tasmota32/release/tasmota32.bin

But it worked with v12.5.0 binary:
https://ota.tasmota.com/tasmota32/release-12.5.0/tasmota32.bin

ESP Flasher is very good, any chances to keep working with tasmota v13?

Jason2866, 2023-11-01:
Can not reproduce. 
[image]

Jason2866, 2023-11-01:
Ahh, wait. When using the version from the OTA Server it bootloops. Will investigate if this can be solved

Jason2866, 2023-11-02:
If i find some spare time i will do a new version which supports the Tasmota safeboot scheme.

felipeapp, 2023-11-02:
> After digging a bit more. The bootloader is still compatible. The answer is plain simple. Most actual Tasmota32 firmwares are too big to fit in the classic two OTA partition scheme (which is still used with ESP Flasher).
> For flashing this bigger firmwares the Tasmota safeboot Partition scheme is needed which ESP Flasher does NOT support.
> For flashing this bigger firmwares use Tasmota Web Installer https://tasmota.github.io/install/

Yes, the web installer works. However, use ESP Flasher to directly flash my custom tasmota firmwares would be great.

Thank you for your time, I'm going to wait for news.

Jason2866, 2023-11-02:
You can try the version from the branch `safeboot`. Tested with an ESP32 and it did work.

felipeapp, 2023-11-02:
> You can try the version from the branch `safeboot`. Tested with an ESP32 and it did work.

Great, this version works, thank you!

Jason2866, 2023-11-02:
Don't be surprised when it breaks in between. It is not final... Will do a release on PyPi when it's done.

Jason2866, 2023-11-03:
Prebuilt binaries in release section and available from PyPi

felipeapp, 2023-11-04:
> Prebuilt binaries in release section and available from PyPi

Great! Everything is woking.

closed 2023-11-05 by Jason2866