Jason2866/ESP_Flasher #29: [FR]: More auto baud rate speeds

alexdelprete, 2023-11-15:
Hi Jason, thanks for this flasher tool, it works beautifully.

I use many Atom Lite devices, which support these serial speeds:

[image]

Before using your program, I was using esptool cli with 1.5Mbps setting: it was fast and reliable on those devices.

`esptool.exe -b 1500000 write_flash 0x0 tasmota32-core2.factory.bin`

This is the log from ESP-Flasher: it would be really nice if you could test more speed rates to autodetection, or have an option to choose Auto, vs manual speedrate.

```
Using 'COM3' as serial port.
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
Connecting....

Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32-PICO-D4 (revision v1.0)
 - Number of Cores: 2
 - Max CPU Frequency: 240MHz
 - Has Bluetooth: YES
 - Has Embedded Flash: YES
 - Has Factory-Calibrated ADC: YES
 - MAC Address: E8:9F:6D:09:63:C0
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Chip does not support baud rate 460800, changing to 115200
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
Connecting....
Uploading stub...
Running stub...
Stub running...
 - Flash Size: 4MB
 - Flash Mode: dio
 - Flash Frequency: 80MHz
Erasing flash (this may take a while)...
Chip erase completed successfully in 2.7s
```

Jason2866, 2023-11-15:
Let me do some tests. My experiences are different, when the speed 460800 is already failing higher speeds never worked. Maybe the ESP32-PICO-D4 is different. Iirc i have one in my drawers. Will do tests. But when only PICO-D4 is supporting the higher speed(s) reliable i will not change the code. It is too specific.

alexdelprete, 2023-11-15:
> My experiences are different, when the speed 460800 is already failing higher speeds never worked

With the Atom Lite I am flashing with esptool via cli at 1.5Mbps without issues.

Anyway, thanks for taking the time to think about this, the Atom Lite is very widespread, it would be cool to have an efficient GUI flashing tool for users.

> The CLI already supports setting ths flash speed.

If in CLI, I always use the official esptool. But I hadn't found a reliable ESP8266/ESP32 GUI flashing tool before.

Thanks Jason.

Jason2866, 2023-11-15:
Did some tests. The updated stub loaders seems improved a lot. Most test devices worked with 1500000 but sadly two get stuck with this high speed and do not fall back to the lower speed. So sadly to say i see currently no way to improve speed without breaking the overall function of the flasher

Jason2866, 2023-11-15:
If you use the python version, you can simply change the starting high speed. Just change this value https://github.com/Jason2866/ESP_Flasher/blob/C2_test/esp_flasher/__main__.py#L38

alexdelprete, 2023-11-15:
> Mhh. After a PC restart the all are flashing with 1500000. Let me do some more test.

No problem. 750000 is also a good speed to test. Works fine here.

> If you use the python version, you can simply change the starting high speed. Just change this value

Unfortunately I have a lot of users/friends that are not experts, use windows, and prefer GUI tools vs the CLI. I mostly use CLI esptool via scripts and aliases. :)

Jason2866, 2023-11-15:
This evening i will release a version with 1.5M. 
Esp8266 was not working. -> Set speed of 115200 as fix value for the esp8266.

alexdelprete, 2023-11-15:
> This evening i will release a version with 1.5M.

Great! Thank you.

> Esp8266 was not working. -> Set speed of 115200 as fix value for the esp8266.

As expected, I think these speed rates are only for ESP32 devices.

Jason2866, 2023-11-15:
No, no speed in between. When 1.5M fails 750k is failing too. As far as i can tell after my tests. So directly fall over to 115200 is imho the only solution which makes sense.
The release will be "just" an pre release, so feedback welcome.

Jason2866, 2023-11-15:
@alexdelprete Give it a try https://github.com/Jason2866/ESP_Flasher/releases/tag/v.2.0.1

alexdelprete, 2023-11-15:
> @alexdelprete Give it a try https://github.com/Jason2866/ESP_Flasher/releases/tag/v.2.0.1

Worked fine, blazing fast! :)

One issue I just found out, not related, it happened with previous release too: when flashing tasmota `firmware.factory.bin` it throws an error. With `firmware.bin` everything ok.

```console
Using 'COM3' as serial port.
Connecting......
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
Connecting.....

Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32-PICO-D4 (revision v1.0)
 - Number of Cores: 2
 - Max CPU Frequency: 240MHz
 - Has Bluetooth: YES
 - Has Embedded Flash: YES
 - Has Factory-Calibrated ADC: YES
 - MAC Address: E8:9F:6D:09:63:C0
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 1500000
Changed.
 - Flash Size: 4MB
Unexpected error: The firmware binary is invalid (magic byte=FF, should be E9)
```

alexdelprete, 2023-11-15:
And also...right after flashing, if I click flash again, it throws this error: I have to restart the app to flash again.

[image]

Jason2866, 2023-11-15:
By design. Tasmota Esp_Flasher is fetching all files needed to flash Tasmota with the safeboot OTA partition scheme. So no factory file needed and not supported.

alexdelprete, 2023-11-16:
Works fine, amazing speed. The message for factory bin images is very clear now.

Only two issues left: the version in the title bar is still 2.0 and then this error message when you click Flash again right after a flashing operation completed:

```
Using 'COM3' as serial port.
Unexpected error: could not open port 'COM3': PermissionError(13, 'Access is denied.', None, 5)
```

Jason2866, 2023-11-17:
Release https://github.com/Jason2866/ESP_Flasher/releases/tag/v.2.2.1
Now supporting Tasmota factory images too.
The COM port error is a weird issue. Needs a very deep dive. Dont know if it is possible to solve.
The behaviour on my MAC is different than with Windows...

alexdelprete, 2023-11-17:
> Now supporting Tasmota factory images too.

Awesome. Now we have the best UI flasher for ESPs. :)

Thanks a lot for accepting the requests Jason, I appreciate it, and my friends will be happy to know they won't have to use CLI.

Regarding the COM port error, it's a niche case, so there's no rush solving that one.

Thanks again.

closed 2023-11-17 by alexdelprete