Jason2866/ESP_Flasher #36: Error while retrieving firmware file

rolanddoros, 2023-12-06:
Using 'COM3' as serial port.
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting...
Detecting chip type... ESP32
Connecting...

Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32-D0WDQ5-V3 (revision v3.1)
 - Number of Cores: 2
 - Max CPU Frequency: 240MHz
 - Has Bluetooth: YES
 - Has Embedded Flash: NO
 - Has Factory-Calibrated ADC: YES
 - MAC Address: 24:DC:C3:C2:9D:58
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 1500000
Changed.
 - Flash Size: 4MB
Unexpected error: Error while retrieving firmware file 'https://raw.githubusercontent.com/Jason2866/ESP_Flasher/factory/bootloader/esp32/bin/bootloader_dout_40m.elf': 404 Client Error: Not Found for url: https://raw.githubusercontent.com/Jason2866/ESP_Flasher/factory/bootloader/esp32/bin/bootloader_dout_40m.elf

Jason2866, 2023-12-06:
Tasmota does not use mode dout for esp32

rolanddoros, 2023-12-08:
With a factory binary it worked thank you.

However, i'm about to tasmotize a zb-gw03 with Ethernet on it. The binary that i was able to flash does not support the ethernet port.

I've seen you have some tasmota binaries, can you suggest one that supports ethernet as well on zb-gw03 using for Zigbeebridge.

rolanddoros, 2023-12-13:
It works! Thank you!

closed 2023-12-13 by rolanddoros