Jason2866/ESP_Flasher #37: 404 when running

DeflateAwning, 2023-12-17:
Trying to install the Tasmota firmware. Got a 404 error within the log output of this program:

```
...
 - Flash Size: 4MB
Unexpected error: Error while retrieving firmware file 'https://raw.githubusercontent.com/Jason2866/ESP_Flasher/factory/bootloader/esp32/bin/bootloader_dout_40m.elf': 404 Client Error: Not Found for url: https://raw.githubusercontent.com/Jason2866/ESP_Flasher/factory/bootloader/esp32/bin/bootloader_dout_40m.elf

```

### Details
* Windows 10
* Installed via pip

Jason2866, 2023-12-17:
Which Tasmota Firmware? Downloaded from? Please provide a link.

DeflateAwning, 2023-12-17:
On mobile, but from here: https://ota.tasmota.com/tasmota/release/

It's this one: https://ota.tasmota.com/tasmota/release/tasmota-4M.bin

And also this one: https://ota.tasmota.com/tasmota/release/tasmota-4M.bin

Flashing onto a board with 4M of memory

Jason2866, 2023-12-17:
Ahh, an esp8266 firmware. That's a bug. For esp8266 no additional files are needed.

DeflateAwning, 2023-12-17:
> Ahh, an esp8266 firmware. That's a bug. For esp8266 no additional files are needed.

Well it supports both ESP8266 and the ESP32. I was flashing onto an ESP32

Jason2866, 2023-12-17:
tasmota-4m is for esp8266! You can't use this firmware for ESP32.
For esp32 you have to use the Tasmota versions with "32" in the name.
Anyways it is a bug, for esp8266 no additional files are needed.
It is fixed now. Install latest version from PyPi

closed 2023-12-19 by Jason2866