Jason2866/ESP_Flasher #44: Error in Debian testing

HombreTriton, 2024-04-23:
Hi,

When I select "browse" the app crash

./ESP-Flasher 
/usr/lib/x86_64-linux-gnu/gvfs/libgvfscommon.so: undefined symbol: g_task_set_static_name
Failed to load module: /usr/lib/x86_64-linux-gnu/gio/modules/libgvfsdbus.so

(ESP-Flasher:44296): Gtk-WARNING **: 19:24:09.212: /lib/x86_64-linux-gnu/libibus-1.0.so.5: undefined symbol: g_task_set_static_name

(ESP-Flasher:44296): Gtk-WARNING **: 19:24:09.212: Loading IM context type 'ibus' failed

(ESP-Flasher:44296): Gtk-WARNING **: 19:24:09.244: /lib/x86_64-linux-gnu/libibus-1.0.so.5: undefined symbol: g_task_set_static_name

(ESP-Flasher:44296): Gtk-WARNING **: 19:24:09.244: Loading IM context type 'ibus' failed

(ESP-Flasher:44296): Gtk-WARNING **: 19:24:09.244: /lib/x86_64-linux-gnu/libibus-1.0.so.5: undefined symbol: g_task_set_static_name

(ESP-Flasher:44296): Gtk-WARNING **: 19:24:09.244: Loading IM context type 'ibus' failed

(ESP-Flasher:44296): Gtk-WARNING **: 19:24:09.245: /lib/x86_64-linux-gnu/libibus-1.0.so.5: undefined symbol: g_task_set_static_name

(ESP-Flasher:44296): Gtk-WARNING **: 19:24:09.245: Loading IM context type 'ibus' failed

(ESP-Flasher:44296): Gtk-WARNING **: 19:24:09.245: /lib/x86_64-linux-gnu/libibus-1.0.so.5: undefined symbol: g_task_set_static_name

(ESP-Flasher:44296): Gtk-WARNING **: 19:24:09.245: Loading IM context type 'ibus' failed

(ESP-Flasher:44296): Gtk-WARNING **: 19:24:09.255: Could not load a pixbuf from icon theme.
This may indicate that pixbuf loaders or the mime database could not be found.
/usr/lib/x86_64-linux-gnu/gvfs/libgvfscommon.so: undefined symbol: g_task_set_static_name
Failed to load module: /usr/lib/x86_64-linux-gnu/gio/modules/libgioremote-volume-monitor.so
/usr/lib/x86_64-linux-gnu/gvfs/libgvfscommon.so: undefined symbol: g_task_set_static_name
Failed to load module: /usr/lib/x86_64-linux-gnu/gio/modules/libgvfsdbus.so
/usr/lib/x86_64-linux-gnu/gvfs/libgvfscommon.so: undefined symbol: g_task_set_static_name
Failed to load module: /usr/lib/x86_64-linux-gnu/gio/modules/libgioremote-volume-monitor.so
**
Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /usr/share/icons/breeze-dark/status/16/image-missing.svg: Unable to load image-loading module: /usr/lib/x86_64-linux-gnu/gdk-pixbuf-2.0/2.10.0/loaders/libpixbufloader-svg.so: /tmp/_MEIXsohUw/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /lib/x86_64-linux-gnu/libicuuc.so.72) (gdk-pixbuf-error-quark, 5)
Bail out! Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /usr/share/icons/breeze-dark/status/16/image-missing.svg: Unable to load image-loading module: /usr/lib/x86_64-linux-gnu/gdk-pixbuf-2.0/2.10.0/loaders/libpixbufloader-svg.so: /tmp/_MEIXsohUw/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /lib/x86_64-linux-gnu/libicuuc.so.72) (gdk-pixbuf-error-quark, 5)
Abortado (`core' generado)

Jason2866, 2024-04-24:
Debian is not supported.

Jason2866, 2024-06-13:
Released Version 3.0.0 should work on all Platforms which are supporting PyQt5
The GUI has been completely rewritten with PyQt5

closed 2024-06-13 by Jason2866