Jason2866/ESP_Flasher #45: esp_flasher error with flash_mode=dout

philippe26, 2024-05-01:
in spi_flasher/common.py
The function read_firmware_info provides up to 4 flash_mode (qio, qout, dio, dout), but only qio and dio bootloaders are provided in esp32/bin (eg bootloader_dio_40m.elf)
I do not know the difference beetwen each flash mode, but using only dio mode seem to run well, so i replace the line 167 of common.py by
         flash_mode = {0: "qio", 1: "qio", 2: "dio", 3: "dio"}.get(flash_mode_raw)

allowing the tasmota upload to complete successfully

Jason2866, 2024-05-01:
yes, i do not provide bootloaders for mode DOUT. It is technically not correct to replace DOUT with the DIO bootloader. If the device does not support DIO it will crash.
The error you have got should never happen when using Tasmota firmwares, since the flash mode DOUT is not used.

s3bs, 2024-05-04:
I've seen the same. This is using Sonoff Zigbee Bridge Pro 2021.11.12 v1.1. 
The error I was making is was using the incorrect firmware. Make sure to use the ESP32 firmware from Tasmota and not anything else. https://ota.tasmota.com/tasmota32/release/

closed 2024-05-04 by Jason2866

philippe26, 2024-05-04:
I've migrated to platformIO instead of standalone esp_flasher python program, and since the migration I didn't exercise the issue. Maybe a firmware error as s3bs said. Thanks for your reply