Jason2866/ESP_Flasher #51: Unexpected error: Error while retrieving firmware file

spiri439, 2024-09-09:
Hello,

Can you please reupload this file https://raw.githubusercontent.com/Jason2866/ESP_Flasher/factory/bootloader/esp32/bin/bootloader_dout_40m.elf

Thank you,

Jason2866, 2024-09-09:
No, since Tasmota setup does not use.

closed 2024-09-09 by Jason2866

CaptClaude, 2024-11-26:
> Hello,
> 
> Can you please reupload this file https://raw.githubusercontent.com/Jason2866/ESP_Flasher/factory/bootloader/esp32/bin/bootloader_dout_40m.elf
> 
> Thank you,

Make sure you are using the correct firmware for your chip. I was about to open another issue with the same error when I realized I was not uploading the correct firmware.  I have an ESP32-S3 and tasmota32s3.bin worked perfectly.