Jason2866/ESP_Flasher #64: Error while retrieving firmware file

francwalter, 2025-08-30:
I tried to use this ESP-Flasher (latest version 3.1.0) to flash a Tasmota 4 MB Firmware on my ESP32 with 4 MB Flash but I get:

`Unexpected error: Error while retrieving firmware file 'https://raw.githubusercontent.com/Jason2866/ESP_Flasher/factory/bootloader/esp32/bin/bootloader_dout_40m.elf': 404 Client Error: Not Found for url: https://raw.githubusercontent.com/Jason2866/ESP_Flasher/factory/bootloader/esp32/bin/bootloader_dout_40m.elf`

So some hardcoded links are broken.
I was able before to flash some OpenDTU firmware with this tool though. 

Thanks. frank

Jason2866, 2025-08-30:
dout mode is not supported. The tool is explicit made for flashing Tasmota.

Jason2866, 2025-08-30:
Defect or wrong firmware used

vytasek, 2025-12-14:
@francwalter flash tasmota32.bin instead of tasmota.bin. Had same issue.